ricaun/ArduinoUniqueID #2: No guarantee for uniqueness on Arduino Nano

c7h, 2019-07-24:
I tested in with a bunch of cheap chinese arduino nanos (Atmega328pb) and 2 of them resulted have the same UUID.

ricaun, 2019-07-24:
That's strange...
I have 10 atmega328pb and all have different IDs.
What board are you selecting?
Could you share the ID, should have 10 bytes!
See ya!

ricaun, 2019-11-15:
I’m closing this issue because it has been inactive for a few months. This probably means that it is not reproducible or it has been fixed in a newer version. If it’s an enhancement and hasn’t been taken on for so long, then it seems no one has the time to implement this.

Please reopen if you still encounter this issue.

Thank you!

closed 2019-11-15 by ricaun

zhekaus, 2021-01-18:
I've got a big package of arduino nano which have the same unique id too. So, I can't relay on this way.

ricaun, 2021-01-18:
Your Arduino nano (ATMEGA328P) could be a counterfeit microcontroller. 

If you are buying from an unofficial seller your chip is not guaranteed is real atmega328p.

https://youtu.be/eeDC1m7ANJI

The library uses the same technique like the video.

zhekaus, 2021-01-19:
Of course these nanos are all counterfeit. Most of arduino in the world are clones. However, they work well.

ricaun, 2021-01-19:
The board yes it's open-source no problem, but the MCU could be counterfeit and that's could be the problem.