ricaun/ArduinoUniqueID #4: How to address Arduino board through the unique ID

Gbouna, 2019-08-06:
Thank you for your library, i have tested it and got unique addresses for Arduino Mega. I have attached the unique address generated by your library, it is about 9 byte. I want to know how i can use the unique ID to address the Arduino. I'm doing a project that has many arduino and they are all controlled by another arduino which acts as a central control. How do i use the unique ID to send message or request data from a particular arduino  board?. 
Thank you

[image]

ricaun, 2019-08-06:
Thanks, man!
I don't know what kind of communication are you using. Usually, I use 4 bytes to make the program simple.
I implement this concept on the [loranow library ](https://github.com/ricaun/loranow) to identify each node automatically.
Basically, I use this...
```
uint32_t id = *(uint32_t *) &UniqueID8[4];
Serial.println(id, HEX);
```
to create a unique 32 bit id.
Then create a simple protocol whos sends this id and check, if the package has the same id.

😄 

What kind of communication are you using?

Gbouna, 2019-08-07:
Thanks for your response. 
I’m using I2C protocol. If you don’t mind could you share with me a complete code either with I2C or any other protocols you have tested and it worked on how to use the unique ID to communicate with a particular Arduino board among many Arduino boards connected to the network. Thank you

Gbouna, 2019-08-07:
Secondly, one of the reason I’m trying to get the unique ID is to find a way of programming all the Arduino connected to the network at once. Basically in my project, all the Arduino connected to the network are doing the same thing but deployed at different locations. So I’m trying to find a way to program them all from another Arduino board which serves like a master. I want to avoid the issue of programming the slaves Arduino one by one. 
Is there any effective way you know how I can go about this? Can you share with me any approach that will work well. 
Thank you

ricaun, 2019-08-07:
Hello, you want a unique code to all the slaves.

You can use something like this...

```
//
// ArduinoUniqueIDWireSlave.ino
//
// Example with UniqueID and Wire Slave.
//

#include <ArduinoUniqueID.h>
#include <Wire.h>

byte id = UniqueID8[7] & 0x7f; // Wire max 127 address, using the last id byte

void setup()
{
  Serial.begin(115200);
  Serial.print("Slave id: 0x");
  if (id < 0x10)
      Serial.print("0");
  Serial.println(id, HEX);
  Wire.begin(id);
  Wire.onRequest(requestEvent);
}

void loop()
{
}

void requestEvent() {
  Wire.write(id);
}
```

The I2C has a limitation with 127 max address, in this case, the code is using the last 7 bits, of the id... If you have bad luck you can have 2 devices with the same address.

You can use some input to add one to the address, like some I2C chips, and fix this same address problem.

You can use [I2cScanner](https://playground.arduino.cc/Main/I2cScanner/) on the master to find all the address connected.

Gbouna, 2019-08-08:
Thanks for your response again, that was great. 
One of the reason I’m trying to get the unique ID is to find a way of programming all the Arduino connected to the network at once. Basically in my project, all the Arduino connected to the network are doing the same thing but deployed at different locations. So I’m trying to find a way to program them all from another Arduino board which serves like a master. I want to avoid the issue of programming the slaves Arduino one by one. 
Is there any effective way you know how I can go about this? Can you share with me any approach that will work well. 
Thank you

ricaun, 2019-08-21:
I don't know how many Arduino are you using and what kind of network are you trying to create... But another way is using some jumpers to configure the id, but you need some io's free to use, could be a problem.
Or you can save some id on the eeprom using the serial monitor, but you still need to connect one by one and configure.

ricaun, 2019-11-15:
I’m closing this issue because it has been inactive for a few months. 

Thank you!

closed 2019-11-15 by ricaun