ricaun/ArduinoUniqueID #6: Not unique on Mega2560

haimiko, 2020-07-11:
After flashing a few dozen Mega2560 I stumbled on duplicate serial numbers.

ricaun, 2020-07-12:
Hello @haimiko 

The Mega2560 does not have an official serial number on the datasheet. 

So a repeat serial number could be should occur... 😞 

Could you try the code below and share the result, the code shows all the signature bytes of the mcu on the serial monitor.

```
#include <avr/boot.h>

#ifndef SIGRD
#define SIGRD 5
#endif

void setup() {
  Serial.begin(115200);
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial.println("");
  for (int i = 0x00; i < 0x40; i++ ) {
    if ((i % 0x10) == 0)
      Serial.println();
    byte b = boot_signature_byte_get(i);
    
    if (b < 0x10)
      Serial.print("0");  
    Serial.print(b, HEX);
    Serial.print(" ");
  }

  Serial.println("");
  for (int i = 0x00; i < 10; i++ ) {
    byte b = boot_signature_byte_get(0x0E + i);
    if (b < 0x10)
      Serial.print("0");  
    Serial.print(b, HEX);
    Serial.print(" ");
  }
  Serial.println("");
  delay(5000);
}
```

I appreciate if you could share the boards with the same serial number.

Thanks

haimiko, 2020-07-12:
Only encountered 2 duplicates so far.  Here is the output:

**1st Mega2560:**  returned serial #: 6E756E6B776F02C
```
1e 9c 98 ff 01 ff ff ff ed ff ff f7 ff ff 6e 75
6e 6b 77 6f ff 00 02 0c 17 02 12 08 13 08 ff ff
ff ff ff ff ff ff ff ff ff ff ff ff 70 1c ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

6e 75 6e 6b 77 6f ff 00 02 0c
```

**2nd Mega2560** returned serial #: 6E756E6B776F02C
```
1e 55 98 ff 01 ff ff ff ed ff ff f7 ff ff 6e 75
6e 6b 77 6f ff 00 02 0c 17 02 12 08 13 08 ff ff
ff ff ff ff ff ff ff ff ff ff ff ff 30 1c ff ff
ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff

6e 75 6e 6b 77 6f ff 00 02 0c
```

Thanks

ricaun, 2020-07-12:
Cool! Looks the same and it's bad... 

You found two chips with the same id 😆 

The code is based on the https://loranow.com/uniqueid/ and this old datasheet of the atmega328pb
[image]

The Device Signature is the same: 1e9801 - Mega2560
RC oscillator Calibration byte: Is different and it's ok!
The rest of the byte looks the same, except the byte `0x002E` with the value `70` and `30` probably is not related to the supposed serial number.

Congratulation! You have two devices with the same `serial`. 

Like I said It's not an official `serial number` and you have the luck and found two mcu really similar.

How many **Mega2560** did you test? Could you share other output, maybe we could use another byte on the serial number o make this odd case different?

I have these two **Mega2560** laying around the lab, and this is the output.

```
1e a5 98 ff 01 ff ff ff ed ff ff f7 ff ff 59 36 
35 30 32 37 ff 10 17 0e 17 08 12 07 13 c0 ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff 70 1c 3f 0c 
a2 26 c3 02 a3 0a 90 01 ff ff ff ff ff ff ff ff 

59 36 35 30 32 37 ff 10 17 0e 
```

```
1e af 98 ff 01 ff ff ff ed ff ff f7 ff ff 59 35 
39 30 30 30 ff 0c 0d 0d 17 08 12 07 13 07 ff ff 
ff ff ff ff ff ff ff ff ff ff ff ff 40 1c b7 0c 
ba 27 e4 02 33 0b d2 01 ff ff ff ff ff ff ff ff 

59 35 39 30 30 30 ff 0c 0d 0d
```

The byte `0x002E` is a good candidate, but in the end, I guess is not unique and a big disclaim saying that is better them trying to make different using some unknown z-address.

Sorry for the long response... 😆 

See yaa!

haimiko, 2020-07-12:
That's what I was afraid of.  As you said the datasheet makes no mention of this.  I've tested about 20 so far and only encountered this once. Thank you for sharing.

haimiko, 2020-07-13:
Just encountered another duplicate.   Mega2560's were not engineered to have unique serial numbers, unlike the mega328pb.

closed 2020-07-13 by haimiko