ricaun/ArduinoUniqueID #7: SAMD51 not supported (and will crash the CPU if used)

baekgaard, 2020-07-23:
Thanks for making this library available.

On the SAMD51 platform, the addresses to use are as follows:

```
#define SERIAL_NUMBER_WORD_0	*(volatile uint32_t*)(0x008061FC)
#define SERIAL_NUMBER_WORD_1	*(volatile uint32_t*)(0x00806010)
#define SERIAL_NUMBER_WORD_2	*(volatile uint32_t*)(0x00806014)
#define SERIAL_NUMBER_WORD_3	*(volatile uint32_t*)(0x00806018)

```

My board is a Feather M4; it can be detected by having ARDUINO_FEATHER_M4 defined, but there may be better/other ways of doing this (not sure if there is a SAMD21 vs SAMD51 definition).

ricaun, 2020-07-24:
Hello, 
The last time I was looking the SAND boards I code this:
```
#if defined (__SAMD51__)
	// SAMD51 from section 9.6 of the datasheet
	#define SERIAL_NUMBER_WORD_0	*(volatile uint32_t*)(0x008061FC)
	#define SERIAL_NUMBER_WORD_1	*(volatile uint32_t*)(0x00806010)
	#define SERIAL_NUMBER_WORD_2	*(volatile uint32_t*)(0x00806014)
	#define SERIAL_NUMBER_WORD_3	*(volatile uint32_t*)(0x00806018)
#else
//#elif defined (__SAMD21E17A__) || defined(__SAMD21G18A__)  || defined(__SAMD21E18A__) || defined(__SAMD21J18A__)
	// SAMD21 from section 9.3.3 of the datasheet
	#define SERIAL_NUMBER_WORD_0	*(volatile uint32_t*)(0x0080A00C)
	#define SERIAL_NUMBER_WORD_1	*(volatile uint32_t*)(0x0080A040)
	#define SERIAL_NUMBER_WORD_2	*(volatile uint32_t*)(0x0080A044)
	#define SERIAL_NUMBER_WORD_3	*(volatile uint32_t*)(0x0080A048)
#endif
```
I almost update but I didn't found a good defined... Some boards/platform uses `__SAMD51__` others not. 😢 

I goona commit this update and we can find a good/better solution.
 
See yaa

baekgaard, 2020-07-27:
Thanks -- yes, that will work; I assume that all SAMD51 platforms have the same set of registers and it appears that also in my setup (under platformio) __SAMD51__ is defined.

ricaun, 2020-08-11:
Perfect, gonna finish the issue and release a new version.

closed 2020-08-11 by ricaun