ricaun/ArduinoUniqueID #8: Writable Unique ID

TinManAkshay, 2020-12-03:
Hi all,

I understand that this library is helping to read the unique ID of the MCU but I wanted to ask theoretically that we can write to the unique ID as well like to change it? Is it possible?

Please let me know. 

Thanks,
AKshay

ricaun, 2020-12-04:
Hello @TinManAkshay 

I believe is not possible to change the id of the MCU, the value is set on the manufacture of the chip and is a serial number. 

If you need to create your own serial number the easiest way is use the EEPROM of the chip and set some values like a serial number.

See yaa!

TinManAkshay, 2020-12-04:
Hi @ricaun ,

Thanks for the response. Are you familiar with writing anything to emulated EEPROM area of SAMD MCUs?

Thanks,
Akshay

ricaun, 2020-12-04:
Never try EEPROM with SAMD MCUs.

I know have some librarys for this cases.

https://github.com/arduino/ArduinoCore-samd/issues/145

See yaa

closed 2020-12-04 by ricaun