ricaun/ArduinoUniqueID #21: getting the same id on different board FF DF FF DF FF FF FF FF

willmendil, 2022-06-06:
Hi, 

Maybe I'm doing something wrong but when trying your code ov pro mini 3V3 I'm always getting FF DF FF DF FF FF FF FF any clue?

ricaun, 2022-06-06:
You probably have a counterfeit atmega328p, here is an excellent video about the subject: https://www.youtube.com/watch?v=eeDC1m7ANJI

[Not Unique Disclaimer](https://github.com/ricaun/ArduinoUniqueID/blob/master/MCU.md#Disclaimer)
_The Atmega328p, Atmega2560, Attiny85 does not have a 'Serial Number' on the datasheet, this mean it's possible to have two microcontroller with the same UniqueId._

willmendil, 2022-06-06:
Having clones is a high probability, as i mainly use ebay for cheap boards for hobby projects.