ricaun/ArduinoUniqueID #24: UNO R4 not supported

GralfR, 2023-08-27:
Loading this lib and selecting the new UNO R4 Minima lists the examples as `incompatible`.
Is there any way to get a unique ID from the UNO R4?
Using ethernet-shields it's very important to have a unique MAC adress. I did this on AVR in code using the unique ID from the board. That makes it easy not to forget generating a "unique" MAC when using more than one ethernet shield.

I read something that native USB support requires unique IDs. The UNO R4 has native USB like many other non-AVR-devices. So maybe there is a "native" arduino command for that?

Any ideas?
Thanks

maxgerhardt, 2023-08-28:
FYI, the 128-bit unique ID for the Uno R4 (Minima, WiFi) and Portenta C33 can be readout using SDK-provided functions
```cpp
#include <Arduino.h>

void setup() {
    Serial.begin(9600);
}

void loop () {
    // 16 bytes / 128 bits
    bsp_unique_id_t const* unique_id = R_BSP_UniqueIdGet();
    Serial.println("Unique ID: ");
    for(int i = 0; i < sizeof(unique_id->unique_id_bytes); i++) {        
        if(unique_id->unique_id_bytes[i] < 0x10)
            Serial.print("0");
        Serial.print(unique_id->unique_id_bytes[i], HEX);
        Serial.print(" ");
    }
    Serial.println("");
    delay(1000);
}
```
Consider this for inclusion in the library.

GralfR, 2023-08-28:
Great job! Thanx! Works perfect on my UNO R4 Minima.

closed 2023-08-28 by GralfR